martin-ger/esp_mqtt #1: Error with Static IP on STA

roboticboyer, 2017-06-17:
Hi Martin
I have tested the new firmware. 
I have set a static IP address for the ESP in the uplink network (e.g. 192.168.1.250), saved and rebooted the ESP.
After this action, the other ESPs, the PC, a Mob have stopped to connect to the AP.
There was the connection but the DHCP service on the ESP_MQTT has stopped to work on the AP side.
FYI in the former esp_uMQTT_broker there wasn't this issue.

As a temporary solution I have reset the use of DHCP with a dynamic IP address for the ESP in the uplink network. 
Now I can again connect the other devises to the AP using DHCP

martin-ger, 2017-06-17:
Hi Giorgio,

tried to reproduce the error error here, did not manage it.
My config:
```
CMD>show
STA: SSID:xxx PW:xxx                                           
AP:  SSID:MyAP PW:none [open] IP:192.168.4.1/24                                 
Static IP: 192.168.178.251 Netmask: 255.255.255.0 Gateway: 0.0.0.0              
MQTT client enabled                                                             
MQTT host: 192.168.178.39                                                       
MQTT port: 1883                                                                 
MQTT user: none                                                                 
MQTT password:                                                                  
MQTT id: ESPBroker_117f13                                                       
Clock speed: 160 
```
My notebook connected at: 192.168.4.2 and was able to connect to the broker:
```
CMD>show stats
System uptime: 0:04:09                                                          
External IP-address: 192.168.178.32                                             
1 Station connected to AP
CMD>show mqttCurrent clients:                                                                
martin_130907526                                                                
Current subsriptions:                                                           
Retained topics:                                                                
MQTT client connected
```
The static IP disables the DHCP client, the server is only disabled, when the AP is disabled. (This is the idea that I tried to put in the code and at least for me that seems to work). Did you also set netmask and gateway? (Should have no impact on the DHCP server, but anyway...)

In your setup, when a client connects to the AP (should be logged at serial port of the ESP), it does not get any DHCP config?

roboticboyer, 2017-06-17:
Hi Martin
I have not set the gateway on the STA, only the fix IP
the clients are connected to the AP but the dhcp dosn't assign any IP

TY&BR
Gio

martin-ger, 2017-06-17:
Could you send me the results of "show" and "show stats"?

roboticboyer, 2017-06-17:
I have set the netmask
`set netmask 255.255.255.0`
Now the AP clients receive the IP address by DHCP
Tomorrow I'll post further information

roboticboyer, 2017-06-18:
The logs: 

`CMD>show
STA: SSID:Network_2 PW:password
AP:  SSID:MyAP PW:none [open] IP:192.168.4.1/24
Static IP: 192.168.1.250 Netmask: 255.255.255.0 Gateway: 0.0.0.0
MQTT client disabled
Clock speed: 80

CMD>show stats
System uptime: 0:00:27
External IP-address: 192.168.1.250
1 Station connected to AP

CMD>show stats
System uptime: 0:07:46
External IP-address: 192.168.1.250
1 Station connected to AP

CMD>show mqtt_broker
Current clients:
gio, EspGiorgio
Current subsriptions:
EspGiorgio: "/led" (QoS 0)
gio: "#" (QoS 0)
gio: "temperatura" (QoS 0)
gio: "umidita" (QoS 0)
gio: "Trugiada" (QoS 0)
Retained topics:
MQTT client disconnected
`

roboticboyer, 2017-06-18:
`ddd
ssss`

martin-ger, 2017-06-18:
Looks good, doesn't it? Maybe the request is wrongly routed, when Thema netmask matches to anything? Will test this later...

roboticboyer, 2017-06-18:
Yes it's OK! Now it's working.
Must I set the gateway such as 192.168.1.1?

martin-ger, 2017-06-18:
Only, if the ESP should connect to another broker outside the own network. It is the router to the internet.

closed 2017-06-18 by martin-ger