martin-ger/esp_mqtt #2: Internet routing

HynekBaran, 2017-09-05:
Hi,
I would be useful to allow internet routing, as your esp_wifi_repeater do (possibly as a bridge, no NAS). Is it a problem to add such a feature?
Thanks
Hynek Baran

martin-ger, 2017-09-05:
No real problem. It just has to be linken against my version of the esp_open_lwip library instead of the stadard one, e.g. from https://github.com/martin-ger/esp_wifi_repeater/blob/master/liblwip_open_napt.a . It includes routing already.

ShravanNAgni, 2017-11-03:
Hi, 
I flashed the firmware in the firmware directory onto an esp8266(Wemos D1) using ESP8266 Download Tool and if I use the terminal tools like Putty or Hyperterminal, I am unable to write any script.
Tried with both SPI modes QIO and DIO.
Flash size 32Mbit. Baud rate - 230400
I am getting some continuous symbols on the terminal window and the led on esp is flashing continuously.
It would be great if get some help regarding this.
I am using a windows 10 machine,

martin-ger, 2018-02-18:
https://github.com/martin-ger/esp_mqtt#known-issues

closed 2018-02-18 by martin-ger