martin-ger/esp_mqtt #3: Size

scargill, 2017-09-15:
Hi

This looks great but... 8 clients?  I think my little IOT setup has about 15 ESP8266s talking to the broker - any chance of increasing that number or porting to the (now much cheaper than a few months ago) ESP32??

martin-ger, 2017-09-15:
Hi,

sorry, 8 WiFi clients for the SoftAP and max. 10 TCP connections are hard limits in Espressif's ESP8266 SDK code. In addition each MQTT connection needs some KB buffer space. This imposes another limit. Could be reduced, as messages are usually small. But again: 8-10 is the absolute limit. Connecting two brokers and bridging between them might solve a small problem in some cases.

Porting to the ESP32 might be an option, but there are again limits on the number of clients as far as I understood. And I just learned from you that the ESP32 has become significantly cheaper (cheaper than the RasPi ZeroW, which is surely the better plattform for a full broker (Linux+Mosquitto)).

martin-ger, 2017-10-12:
With new queue config, up to 13 clients (in STA mode) can connect. A common out queue might even save more RAM...

closed 2017-10-12 by martin-ger