martin-ger/esp_mqtt #4: enable PWM function?

kim-rok, 2017-09-28:
What a nice project!
my project need PWM control. So how can I enable PWM function?
And where can I get the documents  about scripting language used in "script.pir" and "script.sonoff"?

martin-ger, 2017-09-28:
Sorry, currently there is no more written documentation on the scripting language available than the README. Once consolidated, I will work on this.

Also, there is no buildin PWM support. Guess, it would need an additional pin definition command (like "pinmode input") and a new output action (like "gpio_out"). Both canbe added to"lang.c" (see line 839 ff.). Will think about this for the next release.

martin-ger, 2017-09-28:
Added PWM: Try "gpio_pwm <pin> <val>" in a script. <pin> is the GPIO number, <val> a value between 0 and 1000. There is no need to initialize the pin before, but it cannot be used any more for interrupts or binary i/o in the same script.

martin-ger, 2017-09-29:
Added a sample in the "scripts" directory.

martin-ger, 2017-09-29:
Added a decription of the scripting language here; https://github.com/martin-ger/esp_mqtt/blob/master/SCRIPTING.md

closed 2017-09-29 by martin-ger