martin-ger/esp_mqtt #5: "wdt reset"

kim-rok, 2017-10-03:
After flashing ESP, broker startting message displayed.
And then configure loaded.
after a moment, ESP go to 'wdt reset'.

Any help?

Thankyou for PWM!

martin-ger, 2017-10-03:
Happens sometimes for me with unstable power supply. Good power supply and a larger capacitor between GND and Vcc may help.
Or only with this firmware version? If you download the FW binary make sure you don't corrupt the file (e.g. ASCII mode).

Gruffnco, 2017-10-05:
Hi 
I used the bin files to write to a Wemos d1 using nodemcu master flasher.
Everything went ok but when I reset the wemos I get this:-

-----------------------------------------------

No config found, saving default in flash
Loading default configuration
Saving configuration

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 32308, room 16 
tail 4
chksum 0xcc
load 0x3ffe8000, len 2044, room 4 
tail 8
chksum 0xc4
load 0x3ffe8800, len 6848, room 0 
tail 0
chksum 0xae
csum 0xae
Œãäƒgì“r‚òg|ìlldd`Œãs›dŒlýC¡Hè•¥?Router/MQTT Broker V2.0 starting

Config found and loaded

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 32308, room 16 
tail 4
chksum 0xcc
load 0x3ffe8000, len 2044, room 4 
tail 8
chksum 0xc4
load 0x3ffe8800, len 6848, room 0 
tail 0
chksum 0xae
csum 0xae
Œãäƒgì’r‚òg|ìlddd`Œãs›lŒlýC¡Hè•¥?Router/MQTT Broker V2.0 starting

Config found and loaded

------------------------------------------

This just continues to loop.
Have I done some think wrong?

Gareth

martin-ger, 2017-10-05:
No real idea, for me the latest FW seemed to work fine. You might try an earlier version, perhaps this one https://github.com/martin-ger/esp_mqtt/tree/dd4cf9c2da1c9e2fd2b6c26a3192005be40ebaee - this is known to work fine for others. Currently, I'm on a trip. Will test, if I can find any problem with the latest build on sunday.

Gruffnco, 2017-10-06:
Thanks for the reply Martin

I uploaded the older version from your link and had similar result to the newer firmware:-

------------------------------------------

WiFi Router/MQTT Broker V2.0 starting

Config found and loaded

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 32008, room 16 
tail 8
chksum 0xf0
load 0x3ffe8000, len 2044, room 0 
tail 12
chksum 0x23
ho 0 tail 12 room 4
load 0x3ffe8800, len 9800, room 12 
tail 12
chksum 0x00
csum 0x00
Œãäƒgì“s‚òg|ällll`Œã{›d„ÔÕ

WiFi Router/MQTT Broker V2.0 starting

Config found and loaded

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 32008, room 16 
tail 8
chksum 0xf0
load 0x3ffe8000, len 2044, room 0 
tail 12
chksum 0x23
ho 0 tail 12 room 4
load 0x3ffe8800, len 9800, room 12 
tail 12
chksum 0x00
csum 0x00
„ãìƒoä›{ƒón|ìllll`„âr“lŒÔÕ

WiFi Router/MQTT Broker V2.0 starting

Config found and loaded

----------------------------------------------

I then loaded your esp_wifi_repeater to the same wemos and that worked ok, so I know the wemos is good and the way I upload the firmware is good to.

Not sure what to do now, Hope you can help when you return.

Regards
Gareth

martin-ger, 2017-10-07:
Another idea what might be wrong: perpaps the flash read of the script fails, beause there is something in the memory from previous firmwares. Clould you try to erease flash memory at 0xd000 by e.g. flash this empty bin there:
https://github.com/CHERTS/esp8266-devkit/blob/master/Espressif/ESP8266_SDK/bin/blank.bin

Will also try to fix this, when I am back tomorrow.

Gruffnco, 2017-10-07:
Thanks for that Martin, writing the blank.bin file has sorted my problem. 

Keep up the good work, you have some great projects.

Regards
Gareth

closed 2017-10-08 by martin-ger