martin-ger/esp_mqtt #7: make issue

kim-rok, 2017-10-06:
With the problem "wdt reset", I try remake the firmware.
Before "make" command, I changed the value of the "build area" in the Makefile, and wifi AP mode variable( ssid, and password) in "user_config.h" in user directory.
After correction, I applied "make" command at "user" directory but I got error....

kim@kim-R510-P510:~/opt/esp_mqtt/user$ make
LD build/app.out
/home/kim/opt/esp-open-sdk/sdk/lib/libmain.a(app_main.o): In function `flash_data_check':
(.irom0.text+0x770): undefined reference to `user_init'
/home/kim/opt/esp-open-sdk/sdk/lib/libmain.a(app_main.o): In function `flash_data_check':
(.irom0.text+0xb27): undefined reference to `user_init'
collect2: error: ld returned 1 exit status
../Makefile:122: recipe for target 'build/app.out' failed
make: *** [build/app.out] Error 1

How can I solved the problem?

martin-ger, 2017-10-06:
Try the "make" in the top directory.

kim-rok, 2017-10-06:
Ok well made firmware.

But "wdt reset" continue with big capacitor.
May be need Watchdog time off?

martin-ger, 2017-10-06:
No, should work fine with watchdog. At least it does for me. Please try to disable "SCRIPTED" in user_config.h by commenting it out. Does this change anything?

martin-ger, 2017-10-07:
Another idea what might be wrong: perpaps the flash read of the cript fails, beause there is something in the memory from previous firmwares. Clould you try to erease flash memory at 0xd000 by e.g. flash this empty bin there:
https://github.com/CHERTS/esp8266-devkit/blob/master/Espressif/ESP8266_SDK/bin/blank.bin

Will also try to fix this, when I am back tomorrow.

kim-rok, 2017-10-07:
1) 1MB ESP8266 + firmware with script function + blank.bin flashing,  working AP, but mqtt client connection not  maintained.
2)4MB ESP8266 + firmware with script function + blank.bin flashing, serial consol connected with 74800(not 115200)  and persistent reset(reset code 2) during boot.

kim-rok, 2017-10-07:
solved problem
the cause is unknown problem of esptool and blank.bin file size.
Now sripting running well.
But Mqtt client connection problem happened.

martin-ger, 2017-10-08:
What kind of connection problem?

kim-rok, 2017-10-09:
WiFi connection to MyAP is OK.
But I can not connect to broker using android mqtt client program(broker URL(ap-ssid), port(1883), user(option),password(ap-password) set up).

When wifi connection 
    add  1
    aid  1
    station:mac add  join, AID=1
    station:amc add join, AID =1

display.

When I use mqtt client, my setting is wrong?
Please let me know the method of client setting for mqtt broker.

kim-rok, 2017-10-09:
I figure out the connection problem.
Thank for your support.
Next I will drill the scripting.

My furnace project need PWM, ADC, DHT sensor reading and the result  publish to client.
The client number in my project is only 3. 
Is it possible to add  DHT, ADC modules?
If it is possible, can you explain the method?

martin-ger, 2017-10-09:
PWM is already included, ADC could and should be included soon, but the DHT sensor is hard to add to a basic scripting language as it uses a specific serial bus protocol. Thus, probably the easiest way would be to use the Arduino environment with the DHT library, write a (simple) Arduino program and use esp_uMQTT_broker as an additional library (sample see at: https://github.com/martin-ger/esp_mqtt/blob/master/Arduino/uMQTTBrokerSample.ino )

closed 2017-10-10 by martin-ger

martin-ger, 2017-10-12:
ADC support has been added. ;-)