martin-ger/esp_mqtt #8: Compiling for Arduino

necrolyte, 2017-10-06:
I cannot seem to follow your instructions on the README to understand where mqtt_server.h is supposed to go. I keep getting file not found when compiling. When I've linked directly to that file, I get a 'cannot find user_interface.h. Everything else seems to be in the correct directories.

Can you break this down a little easier for someone like me? Thank you.

necrolyte, 2017-10-06:
Just realized I didn't have the ESP8266 hardware set up correctly. I now understand.

I was working on ESP32, and trying to port. Any help is appreciated.

martin-ger, 2017-10-07:
If you want to port this to ESP32, we should start with the mqtt directory and the C SDK. The Arduino integration is just a minimal hack to use the C lib.

Sorry, I havn't found the time so far to get familiar with the ESP32 while I how have on on my desk. And especially for an MQTT broker it is my feeling that it is still too expensive: a PiZeroW with Mosquitto is much more complete at about twice the costs...

closed 2017-10-26 by martin-ger