martin-ger/esp_mqtt #9: error in make

kim-rok, 2017-10-19:
................
/home/kim/opt/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: build/app.out section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
Makefile:122: recipe for target 'build/app.out' failed
make: *** [build/app.out] Error 1

I got a error with updated data during make. How can I fix it?

kim-rok, 2017-10-19:
I figure it out.
I got the error when the  SSL option enabled.

martin-ger, 2017-10-19:
I think, the code will become to big, if you switch on all options - if you switch off SCRIPED (if you don't need it), it compiles. However, I did not make real tests with the SSL option enabled - it should be considered "experimnntal".

kim-rok, 2017-10-19:
When using the precompiled firwmware, typing the script language is hard because of scrolling of screen with wifi reconnect(unadopted SSID name). 
How do you think about the precompiled firmware with wifi auto-connect option off.
The user can activate the auto connection  with script after flashing and reset.

martin-ger, 2017-10-19:
Good idea! Added that. autoconnect 0 it factory default. "autoconnect 1" will be automatically set, when an ssid is defined with "set ssid". Thus, when you give an SSID and save and reset that, it will start trying to connect (and print to the console on failure).

martin-ger, 2017-10-20:
Found the problem with the ".text' will not fit in regioniram1_0_seg'" - actually I had forgotten to declare a function "ICACHE_FLASH_ATTR" - this filled up the IRAM. Now it works...

closed 2017-10-26 by martin-ger