martin-ger/esp_mqtt #10: not stable

simoncheen, 2017-10-25:
hello,
the mqtt borker become not stable after enable the MQTT_server_onAuth,
once the client disconnected, the system will restart.
here is the logs:

```
Connecting to chan's house
..
WiFi connected
IP address: 
192.168.10.251
Starting MQTT broker

username is: admins

received topic 'console' with data 'sss'
received topic 'console' with data 'sss'

Exception (9):
epc1=0x40237ec2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000313 depc=0x00000000

ctx: sys 
sp: 3ffff9c0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffb60:  5720fa7f 00000004 3fff1dc4 40237a37  
3ffffb70:  00000000 3fff2468 3fff23dc 4022eefb  
3ffffb80:  3fffdaf0 00000000 3fff23dc 4022d002  
3ffffb90:  00000220 00000000 00000000 00000000  
3ffffba0:  00000000 00000000 00000000 00000000  
3ffffbb0:  00000000 00000000 00000000 00000000  
3ffffbc0:  00000000 00000000 00000000 00000000  
3ffffbd0:  9115efab 6d1b6545 506b7749 f7c72ca3  
3ffffbe0:  54a91448 d4a1d253 ab1dd0bd 7e673713  
3ffffbf0:  51ee6c81 195d89d9 c44c1358 21d5a378  
3ffffc00:  4000b5f9 00000008 00000038 3ffffce0  
3ffffc10:  00000008 3ffffce0 682c22ad 673a7d4e  
3ffffc20:  5ab69bbd ad5b4f7e b5761552 39cd3c1c  
3ffffc30:  ec7bd841 64000520 fc222ec2 fd1d8ea0  
3ffffc40:  ffc5d31b baf9fc59 9bd3347d abb747f1  
3ffffc50:  b0f50535 9fe0c482 7d73b811 408b66b8  
3ffffc60:  f605df9c 37f82f23 6ada6bb7 bb55cf62  
3ffffc70:  f874f309 991aa1fb bacd7dad 7935b634  
3ffffc80:  d62abc67 ab6e555c 592c56f8 387e3fd4  
3ffffc90:  00000005 00000000 00000020 40100ec2  
3ffffca0:  3ffe8fc5 401042bf 3ffeca50 40100ec2  
3ffffcb0:  40101bdd 3ffeca50 3ffeca28 3ffffd40  
3ffffcc0:  00007fff 08cf71ca 3ffed494 40101dae  
3ffffcd0:  3ffe987c 00000000 00000000 40101dae  
3ffffce0:  00007fff 08cf71ca 401021ee 00000100  
3ffffcf0:  7fffffff 3ffe987c 4021b153 00000001  
3ffffd00:  ffffffff 00000000 3ffe9111 00000008  
3ffffd10:  4021b1a2 3ffec668 3fff0ba4 00020021  
3ffffd20:  4021b2ae 3ffec668 3fff0ba4 3ffec668  
3ffffd30:  00000002 00000000 00000020 40100ec2  
3ffffd40:  3ffe8fc2 4021a27b 3ffec668 3fff104f  
3ffffd50:  00000000 4020d8eb 3fff1e0c 3fff104f  
3ffffd60:  00000000 00000002 00000000 3ffec668  
3ffffd70:  3fff2306 40224210 3fff2306 3fff22cc  
3ffffd80:  3fff101c 40224210 3fff2306 3fff22cc  
3ffffd90:  3fff101c 00000000 0000001f 00000002  
3ffffda0:  3fff03e8 00000003 3fff101c 402244e1  
3ffffdb0:  3ffe8fc5 401042bf 3ffeca28 402244e1  
3ffffdc0:  40101bdd 3ffeca28 00000003 3fff22cc  
3ffffdd0:  3fff22cc 3fff2218 3fff101c 40224730  
3ffffde0:  40103209 00040000 7fffffff 00000000  
3ffffdf0:  00000022 40103206 00040000 40225664  
3ffffe00:  3ffed188 40102132 00000000 401004d8  
3ffffe10:  3fff2218 3fff2214 00002000 4000050c  
3ffffe20:  3fffc278 40101f88 3fffc200 00000022  
3ffffe30:  3ffffe40 00000001 3fff22cc 00000080  
3ffffe40:  402381fe 00000030 0000001c ffffffff  
3ffffe50:  4022ce58 3fff1da4 3fffdcc0 00000014  
3ffffe60:  3ffee240 00000000 00000000 3fff2388  
3ffffe70:  ffffffff 3fffc6fc 0000005c 00000000  
3ffffe80:  3fff121c 3fff0650 3fff235c 00000030  
3ffffe90:  40228a8d 00000000 0000001a 40106acc  
3ffffea0:  402217fa 3fff2218 3fff2214 40221811  
3ffffeb0:  3fff1dc4 3fff2328 00000006 00000001  
3ffffec0:  00000024 00000036 3ffeca28 3fff04d4  
3ffffed0:  3fff2214 3fff04f0 00000000 4022704a  
3ffffee0:  00000001 0104a8c0 00000011 00000000  
3ffffef0:  40237d58 00000011 3ffe9010 00000030  
3fffff00:  002c0000 06e83a2e 515c5803 3fff235c  
3fffff10:  3fff0650 3fff1da4 00000000 4023931b  
3fffff20:  3ffeee50 0000039d 0000039d 4010020c  
3fffff30:  3ffeee50 0000038a 0000038a 4010020c  
3fffff40:  00000000 3ffe8d1f 3fff2344 4010068c  
3fffff50:  3fff0650 3ffe8a40 3fff23dc 40106acc  
3fffff60:  4023767e 3fff121c 3fff2344 40238a16  
3fffff70:  3fff0650 3fff121c 3fff2344 40238bcd  
3fffff80:  4010085e 00000002 00000000 3fff1074  
3fffff90:  40219ee2 3fffdab0 00000000 3fff0678  
3fffffa0:  3fffdc60 00000000 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

Connecting to chan's house
..
WiFi connected
IP address: 
192.168.10.251
Starting MQTT broker

```

martin-ger, 2017-10-25:
Need some more infos. What is the setup? Complete esp_uMQTT_broker with my user_main or just the libmqtt in your code? Do you use scripting? What is the error scenario? AP or STA? Who (re-)connects?

simoncheen, 2017-10-26:
hello,
I'm using the libmqtt and uMQTTBrokerSample, added "MQTT_server_onAuth"  in AP and AP_STA model.
thing is that , my laptop connected to the esp8266 via the esp's AP ssid, and I run a mqtt client in my laptop.  
in my mqtt client:
1. connect to the esp mqtt broker
2. subscribe a topic, 
3. publish a message
4. disconnect the esp mqtt broker

the 8266 is go reboot.
here is the code of mine:
```
/*
 * esp_uMQTT_broker demo for Arduino
 * 
 * The program starts a broker, subscribes to anything and publishs a topic every second.
 * Try to connect from a remote client and publish something - the console will show this as well.
 */

#include <ESP8266WiFi.h>
//#define MQTT_BUF_SIZE   2048
//#define QUEUE_BUFFER_SIZE 4096

#include "mqtt_server.h"

/*
 * Your WiFi config here
 */
char ssid[] = "yea_create_working";  //  your network SSID (name)
char pass[] = "";       // your network password
const char *ap_ssid = "ESPap";
const char *ap_password = "thereisnospoon";

unsigned int mqttPort = 1883;       // the standard MQTT broker port
unsigned int max_subscriptions = 30;
unsigned int max_retained_topics = 30;

void ICACHE_FLASH_ATTR data_callback(uint32_t *client /* we can ignore this */, const char* topic, uint32_t topic_len, const char *data, uint32_t lengh) {
  char topic_str[topic_len+1];
  os_memcpy(topic_str, topic, topic_len);
  topic_str[topic_len] = '\0';

  char data_str[lengh+1];
  os_memcpy(data_str, data, lengh);
  data_str[lengh] = '\0';

  Serial.print("received topic '");
  Serial.print(topic_str);
  Serial.print("' with data '");
  Serial.print(data_str);
  Serial.println("'");
}

bool ICACHE_FLASH_ATTR auth_data_callback(const char* username, const char *password, struct espconn *pesp_conn){
    Serial.println();
    Serial.print("username is: ");
    Serial.print(username);
    Serial.println();
    Serial.println();

   if ((String)username == "admins" &&  (String)password == "admins" ) {
    return true;
   }else{

    return false;
    }
//  if (username == "admins" && password == "admins") {
//    return true;
//  }else{
 // return false;
//  }
  //password = "admins";
  
  
  }

void setup()
{
  Serial.begin(115200);
  Serial.println();
   
  //Serial.print("buff is:");
  //Serial.print(MQTT_BUF_SIZE);
  
  Serial.println();

  // We start by connecting to a WiFi network
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.mode(WIFI_AP); //chose mode
  WiFi.softAP(ap_ssid, ap_password);
  
  //WiFi.begin(ssid, pass);
  
//  while (WiFi.status() != WL_CONNECTED) {
//    delay(500);
//    Serial.print(".");
//  }
//  Serial.println("");
//  
//  Serial.println("WiFi connected");
//  Serial.println("IP address: ");
//  Serial.println(WiFi.localIP());

/*
 * Register the callback
 */
  MQTT_server_onData(data_callback);

/*
 * Start the broker
 */
  Serial.println("Starting MQTT broker");
 
  MQTT_server_start(mqttPort, max_subscriptions, max_retained_topics);
 MQTT_server_onAuth(auth_data_callback);

/*
 * Subscribe to anything
 */
  MQTT_local_subscribe((unsigned char *)"#", 0);
}

int counter = 0;

void loop()
{
  String myData(counter++);

/*
 * Publish the counter value as String
 */
  //MQTT_local_publish((unsigned char *)"/MyBroker/count", (unsigned char *)myData.c_str(), myData.length(), 0, 0);
  
  // wait a second
//  delay(1000);
}

```

and my mqtt client named "mqttfx"

martin-ger, 2017-10-26:
Simon, can confirm this problem - Think, I found the problem.

Could you try with the patched version of the lib, I uploaded right now?

Will do some more test and fix and upload the sources later (up to now I just deleted line 832 in mqtt_server.c)...

simoncheen, 2017-10-27:
it runs very good now, 
what's the problem ?
and can I please ask a addional question?
1. if I wanna to change the MAX_BUFF_SIZE , do I need to recompile the lib ?
2.if I wanna to enable the MQTT client mode, do I need to recompile the lib ?
3. how can I use the "struct espconn *pesp_conn" for printing the client IP in auth_data_callback ?

martin-ger, 2017-10-27:
The problem is a double "espconn_disconnect()". In the NONOS SDK I never got a problem here, but memory management in Arduino seems to be different (better checks?) - anyway I will double check the disconnection sequence for the different cases (client vs broker initiated).

Your questions:
1. Yes, MQTT_BUF_SIZE currently is a compile time constant (what would you need? More or less?)
2. No, it is included in the lib (the complate functionality of tuanpmt's original esp_mqtt) - only C++ headers are missing for Arduino.
3. You need the complete declaration of the "struct espconn" type and then you can access pesp_conn->proto.tcp->remote_ip. (Guess, you get it from user_interface.h - but I will check, whether this works in Arduino)

simoncheen, 2017-10-27:
thank you for your reply.

1) ok, I will recomple the lib myself.  if this  MQTT_BUF_SIZE can be change in arduino , this would better.
2) I can use the  MQTT client mode like tuanpmt's original esp_mqtt? ok
3) I can not use pesp_conn->proto.tcp->remote_ip in arduino.

closed 2017-10-30 by martin-ger