martin-ger/esp_mqtt #12: On ArduinoIDE , how to use mqtt-client API ?

TridentTD, 2017-11-06:
How to  use  mqtt-client API  such as  
function  "MQTT_InitConnection(....)" , "MQTT_InitClient(..)" etc..
on ArduinoIDE ?

Is there any example ?

Thank you.

TridentTD, 2017-11-06:
When I try to use mqtt-client API , it occur some errors  
 that  undefined reference to  espconn_secure_*****  functions

```
Archiving built core (caching) in: C:\Users\Superman\AppData\Local\Temp\arduino_cache_536667\core\core_esp8266_esp8266_d1_mini_CpuFrequency_80,UploadSpeed_921600,FlashSize_4M3M_eab9a2aacd30f374ecf5501ad2a6bb66.a
C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libmqtt.a(mqtt.o): In function `mqtt_tcpclient_recon_cb':

/home/martin/github/esp_mqtt/mqtt/mqtt.c:552: undefined reference to `espconn_secure_disconnect'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libmqtt.a(mqtt.o): In function `mqtt_tcpclient_recv':

/home/martin/github/esp_mqtt/mqtt/mqtt.c:291: undefined reference to `espconn_secure_disconnect'

/home/martin/github/esp_mqtt/mqtt/mqtt.c:400: undefined reference to `espconn_secure_send'

/home/martin/github/esp_mqtt/mqtt/mqtt.c:434: undefined reference to `espconn_secure_send'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libmqtt.a(mqtt.o): In function `mqtt_tcpclient_connect_cb':

/home/martin/github/esp_mqtt/mqtt/mqtt.c:502: undefined reference to `espconn_secure_set_size'

/home/martin/github/esp_mqtt/mqtt/mqtt.c:506: undefined reference to `espconn_secure_connect'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libmqtt.a(mqtt.o): In function `mqtt_get_type':

/home/martin/github/esp_mqtt/mqtt/include/mqtt_msg.h:160: undefined reference to `espconn_secure_set_size'

/home/martin/github/esp_mqtt/mqtt/include/mqtt_msg.h:160: undefined reference to `espconn_secure_connect'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libmqtt.a(mqtt.o): In function `mqtt_dns_found':

/home/martin/github/esp_mqtt/mqtt/mqtt.c:66: undefined reference to `espconn_secure_send'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libmqtt.a(mqtt.o): In function `MQTT_InitClient':

/home/martin/github/esp_mqtt/mqtt/mqtt.c:820: undefined reference to `espconn_secure_set_size'

/home/martin/github/esp_mqtt/mqtt/mqtt.c:821: undefined reference to `espconn_secure_connect'

/home/martin/github/esp_mqtt/mqtt/mqtt.c:837: undefined reference to `espconn_secure_disconnect'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libmqtt.a(mqtt.o): In function `MQTT_InitLWT':

/home/martin/github/esp_mqtt/mqtt/mqtt.c:849: undefined reference to `espconn_secure_send'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board WeMos D1 R2 & mini.

```

martin-ger, 2017-11-06:
The lib with the crypto is missing. Either compile the lib without the MQTT_SSL_ENABLE or try to add -lssl in the linker-line like you did with the -lmqtt.

TridentTD, 2017-11-07:
In the upper error I used the default libmqtt.a (that I thinke MQTT SSL_ENABLE = 1 )
And retry with linker -lmqtt -lssl  ,  errors are as the following.

```
Archiving built core (caching) in: C:\Users\Superman\AppData\Local\Temp\arduino_cache_657177\core\core_esp8266_esp8266_d1_mini_CpuFrequency_80,UploadSpeed_921600,FlashSize_4M3M_eab9a2aacd30f374ecf5501ad2a6bb66.a
C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_crypto_misc.o):(.rodata+0x0): multiple definition of `unsupported_str'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(crypto_misc.o):(.rodata.unsupported_str+0x0): first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_crypto_misc.o): In function `RNG_initialize':

(.text.RNG_initialize+0x0): multiple definition of `RNG_initialize'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(crypto_misc.o):crypto/crypto_misc.c:110: first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_crypto_misc.o): In function `RNG_custom_init':

(.text.RNG_custom_init+0x0): multiple definition of `RNG_custom_init'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(crypto_misc.o):crypto/crypto_misc.c:141: first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_crypto_misc.o): In function `RNG_terminate':

(.text.RNG_terminate+0x0): multiple definition of `RNG_terminate'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(crypto_misc.o):crypto/crypto_misc.c:154: first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_crypto_misc.o): In function `get_random':

(.text.get_random+0x4): multiple definition of `get_random'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(crypto_misc.o):crypto/crypto_misc.c:166: first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_crypto_misc.o): In function `get_random_NZ':

(.text.get_random_NZ+0x8): multiple definition of `get_random_NZ'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(crypto_misc.o):crypto/crypto_misc.c:222: first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_crypto_misc.o): In function `print_blob':

(.text.print_blob+0x0): multiple definition of `print_blob'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(crypto_misc.o):crypto/crypto_misc.c:287: first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_crypto_misc.o): In function `base64_decode':

(.text.base64_decode+0x4): multiple definition of `base64_decode'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(crypto_misc.o):crypto/crypto_misc.c:329: first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_tls1_clnt.o): In function `do_client_connect':

(.text.do_client_connect+0x4): multiple definition of `do_client_connect'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(tls1_clnt.o):ssl/tls1_clnt.c:164: first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_tls1_clnt.o): In function `do_clnt_handshake':

(.text.do_clnt_handshake+0x38): multiple definition of `do_clnt_handshake'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(tls1_clnt.o):ssl/tls1_clnt.c:96: first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_tls1.o):(.rodata+0x54): multiple definition of `ssl_prot_prefs'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(tls1.o):(.rodata.ssl_prot_prefs+0x0): first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_tls1.o): In function `ssl_ctx_new':

(.text.ssl_ctx_new+0x20): multiple definition of `ssl_ctx_new'
[... 211 lines cut ...]
C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(tls1.o):ssl/tls1.c:2384: first defined here

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_tls1.o): In function `DISPLAY_ALERT':

(.text.DISPLAY_ALERT+0x68): multiple definition of `DISPLAY_ALERT'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(tls1.o):ssl/tls1.c:2483: first defined here

[... cut ...]
C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libssl.a(ssl_tls1_svr.o): In function `do_svr_handshake':

(.text.do_svr_handshake+0x24): multiple definition of `do_svr_handshake'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lib\libaxtls.a(tls1_svr.o):ssl/tls1_svr.c:76: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

martin-ger, 2017-11-07:
This is probably because of C-style declarations in includes. You might try to include everything you need from the MQTT-lib as "extern "C" { ... }". But this all is a hack. 

It wasn't my real intention to provide an MQTT client lib for Arduino as there are already good solutions like the "pubsub". Thus, I did just a quick-n-dirty integration of the native C lib to get the broker running in the background (like a 'demon').

TridentTD, 2017-11-07:
OK, already done.  

Thank you.

TridentTD, 2017-11-07:
One more question, 
if I want to the MQTT Broker know client's  ip address,
which variable to receive connected client's ip address?

martin-ger, 2017-11-07:
The struct espconn in the MqttConnectCallback and MqttAuthCallback contains the IP-address:
`ip_addr_t *remote_addr;
remote_addr = (ip_addr_t *)&(pesp_conn->proto.tcp->remote_ip);`

You don't get the client's ID at this interface right now.

closed 2017-11-11 by martin-ger