martin-ger/esp_mqtt #13: Firmware

ShravanNAgni, 2017-11-06:
Hi,
I flashed the firmware in the firmware directory onto an esp8266(Wemos D1) using ESP8266 Download Tool and if I use the terminal tools like Putty or Hyperterminal, I am unable to write any script.
Tried with both SPI modes QIO and DIO.
Flash size 32Mbit. Baud rate - 230400
I am getting some continuous symbols on the terminal window and the led on esp is flashing continuously.
It would be great if get some help regarding this.
I am using a windows 10 machine,

martin-ger, 2017-11-06:
Sometimes there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/bin/esp_init_data_default.bin?raw=true and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

If your downloaded firmware still doesn't start properly, please check with the enclosed checksums whether the binary files are possibly corrupted.

ShravanNAgni, 2017-11-07:
I tried flashing the above .bin file along with other two binary files in the firmware directory but I am not able to get the results. The led on GPIO2 had stopped flashing rapidly but I am still getting garbage value for every key press in the terminal window.
How do I check the binary files?

martin-ger, 2017-11-07:
Compute with "shasum" the SHA1 checksums of the files. Is there only garbage output after a key pressed? Are you using 115200 bps?

ShravanNAgni, 2017-11-07:
Yes whenever I press a key, there is a garbage output. I am using 230400 bps.

martin-ger, 2017-11-07:
That's the Problem: use 115200

ShravanNAgni, 2017-11-07:
Yeah, that was the problem. 
it is working now. Thanks a lot.

closed 2017-11-07 by martin-ger