martin-ger/esp_mqtt #14: About "uint32_t *args" the first parameter of  MqttDataCallback

TridentTD, 2017-11-11:
//-----------------------------------------------------------------------------------------------
typedef void (*MqttDataCallback)(uint32_t *args, const char* topic, uint32_t topic_len, const char *data, uint32_t lengh);
//-----------------------------------------------------------------------------------------------

How to use is this "args" in MQTT Broker's DataCallback? 
Is it  MQTT_ClientCon type?

martin-ger, 2017-11-12:
When you receive the callback from a remote subscription (via the original client lib) you get a pointer to a MQTTclient struct, when you get it from the local broker it is set to NULL.

In some cases it could be the publishing MQTT_ClientCon, but e.g. for retained messages this doesn't work. Also it is not the intention of MQTT and the pubsub model to provide local extra info on the sender of a publication to a subscriber

TridentTD, 2017-11-12:
Thank you .. because of retained messages..

If for normal messages , can I get a pointer to MQTT_ClientCon struct,
and for retained messages, get a pointer to NULL instead?

martin-ger, 2017-11-12:
Sorry, no - it COULD be done - in fact, you get NULL all the time...

closed 2017-11-13 by martin-ger