martin-ger/esp_mqtt #15: MQTT: Out of mem

miolion, 2017-11-14:
Hi,

I think this is very useful project for small IoT devices.
I have found a minor issue when user give NULL password to the broker...

For examble,
> mosquitto_sub -h 192.168.4.1 -d -u "user" -P "" -t hello
> mosquitto_pub -h 192.168.4.1 -u "user" -P "" -t hello -m message

There are always "MQTT: Out of mem" in the debug log...

Jason

miolion, 2017-11-14:
[mqtt_server.c.zip](https://github.com/martin-ger/esp_mqtt/files/1469981/mqtt_server.c.zip)

The attached is the modified of mine ...

SJ

martin-ger, 2017-11-14:
Thanks for the report! This we just have to skip the "if (password != NULL)". Think it left over in the code from former editing.

miolion, 2017-11-14:
Yes, or this could be like this,

		if (password == NULL) {
		    MQTT_WARNING("MQTT: Password invalid\r\n");
		    MQTT_server_disconnectClientCon(clientcon);
		    return;
		}

		clientcon->connect_info.password = (char *)os_zalloc(password_len+1);

The final modification decision will be up to you,

SJ

miolion, 2017-11-15:
I have found out that Martin has resolved this issue.

closed 2017-11-15 by miolion