martin-ger/esp_mqtt #18: PlatformIO flashing?

starkillerOG, 2017-11-24:
I am farly new to programming an ESP8266, so it might be a dump question.
Is it possible to flash this firmware to the ESP using PlatformIO?

starkillerOG, 2017-11-24:
I run PlarformIO within Atom, but I am not verry fimilliar with both.
Is there a more detailed step by step guide how to flash this software to the esp?

martin-ger, 2017-11-24:
No experience (and idea) with PlattformIO, but there is a description at https://github.com/martin-ger/esp_mqtt#building-and-flashing . 

How to connect and setup your ESP8266 heavily depends on your modul (I would recommend a NodeMCU or Wemos D1 Mini). You find lots of tutorial on youtube on how to get the serial connection for any kind of modul - after that you can use the download tool from the link above.

starkillerOG, 2017-11-27:
@martin-ger Thanks for the information.
Can you also build/compile the code with the "ESP8266 Download Tool"?
Or which program can I use to build/compile the changes to the "user_config.h" file such as my routers ssid and pasword?

martin-ger, 2017-11-28:
Top build the software you will never the esp-open-sdk. Have auch look in 
the "build" section of the Readme.

closed 2018-10-19 by starkillerOG