martin-ger/esp_mqtt #19: Storing Config

rebel7580, 2017-12-04:
Hi -
I've been looking into using spi-flash lib to store config info for another project, and have been having trouble figuring out where exactly the 4096 byte sector should be. I looked at your config_flash.c/.h files to see if I could find out what you were using for the 4096 byte block, and it appears you use FLASH_BLOCK_NO * SPI_FLASH_SEC_SIZE  or 0xc *4096 = 0xc000.  I'm not at all familiar with the ESP memory map (and find what's on the web to be very confusing), but this address seems low to me. so I wanted to check with you if I understood this correctly.  I would in fact like to take (most of) your two files and include them in my project as they do exactly what I want. The only issue is I don't fully understand where the config is going, and why.

Thanks for your time.
Ron

martin-ger, 2017-12-06:
You are rigth - I write it to 0xc000 and it is low, but it is between the 0x00000.bin and the 0x10000.bin file, so there is some space and it works for me. You can also write it above your code. Just look at the start address of your bin-file and its length.

rebel7580, 2017-12-07:
I understand. Thanks!

closed 2017-12-07 by rebel7580