martin-ger/esp_mqtt #23: Console with 'script' command.

msillano, 2018-02-03:
My test configuration: Win 10, mosquitto, node-red, Putty.
On Putty console I get:
```    
    CMD>show config
    STA: SSID:FRITZ!Box 3272 PW:cirohomewifi
    AP:  SSID:MQTT_home PW:sonoffbroker IP:192.168.4.1/24
    Network console access on port 7777 (mode 3)
    MQTT broker max. subscription: 30
    MQTT broker max. retained messages: 30
    MQTT broker max. clients: 0
    MQTT client enabled
    MQTT client host: 192.168.178.23
    MQTT client port: 1883
    MQTT client user: none
    MQTT client password:
    MQTT client id: ESPBroker_ 9c593
    MQTT SSL: off
    NTP server: 1.it.pool.ntp.org (interval: 300 s, tz: 1)
    Clock speed: 160
    Serial bitrate: 115200
    CMD>show stats
    System uptime: 0:03:28
    Free mem: 18896
    Interpreter loop: 65982 us
    External IP-address: 192.168.0.53
    0 Stations connected to AP
    NTP synced: 20:24:19
    CMD>echo config       (hops!)

    Invalid Command       (one blak line before message 'Invalid command')
    CMD>
```
All ok. Now I send a 'script delete' (I must do it before any new srcript ) the I download the script.
(It is 'script.sonoff', modifiend ONLY on start config as you see before) 
```
   CMD>script http://192.168.178.23:85/www/sonoff/sonoffscript02.txt
   HTTP request to http://192.168.178.23:85/www/sonoff/sonoffscript02.txt started
   HTTP script download completed (2733 Bytes)
   _
```
Here it stops: no [CMD>] prompt, and I never seen the "Syntax okay" message.
I must press [CR], the prompt shows, and then I can do the 'reset' command.
(continue)

martin-ger, 2018-02-04:
If you type "show script", you see that the script is loaded? Seems so, when looking on thr other topic.

There is no need to delete an old script. Loading a new one automatically deletes the prior.

msillano, 2018-02-04:
Yes, 'show script' shows the new script, i.e. the download works, also without 'Sintax check'.
And  'script delete' is really not required.  
Thanks
m.s.