martin-ger/esp_mqtt #24: Local and remote

msillano, 2018-02-03:
(continue)
I make a simple 'flow' in node red: a catch-all with debug echo, and 2 commands: blink, toggle, off.
[image]
All looks working.
But I don't  see the commands I expect in debug window.
To be more explicit I change 2 lines: 53 and 95 (about). I replace remote payload:
	`publish remote $status_topic $relay_status retained`   with
	`publish remote $status_topic "{Time:" | $timestamp | ",Power:" | $relay_status | "}" retained` 

In debug window i see:
[image]

and on console only
[image]

Maybe I am confused between 'local' and  'remote',  but how to send status to mosquitto?
More:
the lines: `publish local $status_topic $relay_status retained`  looks to me useless.  None uses the 'local' status topic,  the program uses only 'local' commands.
Best regars.
m.s.

martin-ger, 2018-02-04:
I assume 192.168.178.23 is your mosquitto? In the terminology of the project this would be the remote broker. Somewhat strange, that the external IP of the ESP is 192.168.0.53 - different network?

Your setup is: NodeRed publishs and subscribes to the mosquitto and the ESP subscribes also to the mosquitto?

Try to add some dummy remote publication on the status topic right in the "on mqttconnect". Do you see that? 

Try to lock into the published topics on the ESP with  "mosquitto_sub -h <ESP_ip> -t "#"

There is no need for the external mosquitto. You can also try to publish the command topic directly on the ESP.

msillano, 2018-02-04:
The main PC (192.168.178.23) is connected to WiFiRouter via LAN, and 192.168.0.53 is the sonoff via WIFI. (I use a Sonoff-basic). First time this value was done by the router, then I make it fix.

In production i will have many sonoff+esp_matt, using script to implement simple logic 'fail-safe' (e.g. water pump and level switch),  using a central Raspberry (24/7) with mosquitto and a hi-level standard web interface, plus mySQL for logging.

But I fond my mistake: the node-red was linked to 192.168.0.53:1883 (the ESP broker) and not to mosquitto (192.168.178.23:1883) broker. !!!!!!!!
For that I seen local messages and not remote ones.

Very stupid, sorry.
Apologize for your time.
Now messages works as expected. 

Thanks for your patience and I beg pardon for my poor english :(
Best Regards
Marco Sillano