martin-ger/esp_mqtt #25: Telnet Console Timeout

rebel7580, 2018-02-04:
Is there a way to prevent the telnet console from timing out? Somewhat annoying to have to restart the session.

Thanks.

martin-ger, 2018-02-04:
In line 399 of user_main.c you can change the timeout-val.

`espconn_regist_time(pespconn, 300, 1); // Specific to console only`

The manual says:
```
sint8 espconn_regist_time(
        struct espconn *espconn, 
        uint32 interval, 
        uint8 type_flag
)
```
uint32 interval: timeout interval; unit: second; maximum: 7200 seconds

If timeout is set to 0, timeout will be disabled and ESP8266 TCP server will not disconnect if a TCP client has stopped communication. This usage of timeout=0, is deprecated.

closed 2018-03-21 by martin-ger

rebel7580, 2018-03-21:
Sorry I forget to respond to your response so it could be closed.