martin-ger/esp_mqtt #26: Interrupts

msillano, 2018-02-08:
Sorry, i have more problems.
The script.sonoff works well.
But if I add the fallowing code to enable a switch on GPIO 14
```
% The remote switch
on gpio_interrupt 14 nopullup
do
  println "New state GPIO 14: " | $this_gpio
```
The GPIO 0 (on board button)  ceases to work. 
But GPIO 14 works.
Strange.
Some limit on Interrupt use?
Best regards.
Marco

martin-ger, 2018-02-08:
Marco, thanks for the report.

Obviously, I missunderstood the  easygpio_attachInterrupt() API. There is no individual user_arg for each attach - only the last will be handed over to any GPIO interrupt.

Changed the logic for identifying the interrupt source. Think, I works now.

msillano, 2018-02-08:
Thanks Martin
Best Regards
Marco

closed 2018-02-09 by martin-ger