martin-ger/esp_mqtt #27: Script

msillano, 2018-02-09:
Testing scripts I found a unespected behaviour:
This is accepted:
```
       publish remote $status_topic "{Time:" | $timestamp | ",TOGGLE}"  retained
```
But this not:
```
       publish remote $status_topic | "/user" "{Time:" | $timestamp | ",ON}" retained
```
With the message:
       Error (action command expected) at >>"/user "{Time: | $timestamp | ",ON} retained publish 

One more note: 
for a better utilization of 4000 script bytes, maybe it is possible to:
- trim() any line before storage
- reduce any comment line to only starting '%' (so the line numbers are respected) before storage

Best regards
Marco

martin-ger, 2018-02-09:
You are right, for the topic_id I currently do not full expression parsing. A variable instead of a constant works, so it can be dynamic. I could change that in the next future, not too much lines required.

Compression during load would work as well, but it would already require basic parsing before writing to to memory and "show script" would show something more obfuscated. When reading it from flash into RAM for interpretation I already compress it in the way you suggest.

Do you run into problems with script size? 4KB is somewhat arbitrary. Could increase that...

jnherm, 2018-02-17:
May I know what language is the script you are using Martin? THanks!

msillano, 2018-02-17:
It is an 'ad hoc' Martin's script  language, not similar to any other.
For documentation see  [https://github.com/martin-ger/esp_mqtt/blob/master/SCRIPTING.md](https://github.com/martin-ger/esp_mqtt/blob/master/SCRIPTING.md).
Regards

jnherm, 2018-02-18:
Thank you @msillano !

martin-ger, 2018-02-18:
If you need more freedom for programming, I would recommend the Arduino version: https://github.com/martin-ger/uMQTTBroker where you can write the application completely in C++.

closed 2018-02-27 by martin-ger