martin-ger/esp_mqtt #28: Sonoff: using GPIO1 and GPIO3

msillano, 2018-02-10:
One more question:
there is in sp_mqtt a configuration that leaves the GPIO1 (tx) and GPIO3 (rx) totally free ? So they can be used as generic I/O by the user on script?
With Sonoff you do not have many pins available :)

Thanks, regards
Marco

martin-ger, 2018-02-10:
Yes, you can:
Set system_output to 0,this disables all system and script output to serial. Then set the pinmode and use the GPIO commands as usual.

Telnet to port 7777 still gives you access to the CLI.

```
% Config params, overwrite any previous settings from the commandline
config system_output 0

on init
do
	gpio_pinmode 1 input pullup
	gpio_pinmode 3 input pullup

on gpio_interrupt 1 pullup
do
	println "Interrupt GPIO 1: " | $this_gpio

on gpio_interrupt 3 pullup
do
	println "Interrupt GPIO 3: " | $this_gpio
```

msillano, 2018-02-11:
Thanks, that will be useful.

Hope you like see a first result:
https://www.hackster.io/msillano/adding-a-remote-switch-to-sonoff-one-pin-for-switch-led-901cd3

Thanks for the support.
Best regards, 
Marco

martin-ger, 2018-02-11:
Thanks for the project description. Nice device.

And also always love to read about applications of my SW.

Best regards,
Martin

closed 2018-02-18 by martin-ger

msillano, 2018-08-01:
About using GPIO01:
The followig simple scrip

`config system_output 0
on init
do
setvar $t = 0
gpio_pinmode 1 output
gpio_pinmode 14 output
gpio_out 1 0
gpio_out 14 0
settimer 3 3000

on timer 3
do
if $t = 0 then
println "  int timer3 set 1"
gpio_out 1 1
gpio_out 14 1
gpio_out 13 0
setvar $t = 1
settimer 3 200
else
println " int timer3  set 0" 
gpio_out 1 0
gpio_out 14 0
gpio_out 13 1
setvar $t = 0
settimer 3 200
endif `

Works well on GPIO14  and not on GPIO01.
Strange.
Best regarda
Marco