martin-ger/esp_mqtt #29: What is the correct LED behavior?

bilogic, 2018-02-13:
Hi,

I was trying to upload the firmware to a NodeMCU with no success:
- LED stays unlit after burn
- If I press reset, LED will start a blink pattern
- If I unplug the USB cable and replug, LED stays unlit until I press reset

What is the correct LED behavior after a proper upload? 
Thank you.

martin-ger, 2018-02-13:
The LED should be off. You probably ran in into this problem:

Sometimes there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/bin/esp_init_data_default_v08.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

bilogic, 2018-02-13:
THANK YOU! It works and I see MyAP now! I should have written in hours earlier :)
Just curious, how do I read the serial? (You mentioned garbage on the serial)

Once again, thank you!

martin-ger, 2018-02-13:
Just connect with 115200 bps to the UART (NodeMCU USB connection) and use a terminal program (e.g. the Arduino Serial Monitor).

bilogic, 2018-02-13:
Alright, thank you for the help!

closed 2018-02-13 by bilogic