martin-ger/esp_mqtt #31: TCP console for Windows

jnherm, 2018-02-17:
Hello Martin,
May I know how to connect my windows PC to your firmware console using TCP port 7777? I type in my browser 192.186.X.Y:7777 but nothing happens.
Thank you in advance!

Joy

martin-ger, 2018-02-17:
You have to use a terminal program. On Windows "putty" is a perfect choice. Set the IP, protocol "telnet", and Port 7777.

jnherm, 2018-02-18:
Thank you martin-ger!

jnherm, 2018-02-18:
Hello again Martin,
I downloaded PUTTY but then it often closed the connection. What seems to be the problem? By the way what is (mode 3)?

martin-ger, 2018-02-18:
But you get the connection? There is a timeout for the console connection of 5 min of inactivity, that's a feature... ;-)

jnherm, 2018-02-18:
Yes, i got a connection. Then i type in commands, the first 1 to 3 or 4 commands gets a response, but the succeeding commands hangs up. My firewall in OFF, my OS is win XP. By the way, I also use android telnet app, it is more responsive but after not so many commands it hangs up.

martin-ger, 2018-02-18:
Maybe your WiFi connection is a little bit weak? Don't think, there is a known problem.

jnherm, 2018-02-18:
Thank you Martin for the reply. I'll change my wifi router later. By the way, off topic, what will i add on the script so that after restart/brownout relay will restore to previous state?

martin-ger, 2018-02-18:
Store the desired state (0 or 1) in a flash variable (e.g. @1) after each state change. This var will persistently store the value. In "on init" after reboot you then send this state to the GPIO pin again.

jnherm, 2018-02-18:
Thank you again Martin. Anyway, i changed my router and it work fine now. My router should have been the culprit.

jnherm, 2018-02-18:
Another question Martin, how could I change MQTT client host, port, user, password and id from the console? I can't find any "set" command to change this settings.

martin-ger, 2018-02-18:
That are those variables you find under: https://github.com/martin-ger/esp_mqtt#mqtt-clientbridging-functionality

- set mqtt_host IP_or_hostname: IP or hostname of the MQTT broker ("none" disables the MQTT client)
- set mqtt_user username: Username for authentication ("none" if no authentication is required at the broker)
- set mqtt_user password: Password for authentication
- set mqtt_ssl [0|1]: Use SSL for connection to the remote broker (default: 0 = off)
- set mqtt_id clientId: Id of the client at the broker (default: "ESPRouter_xxxxxx" derived from the MAC address)

jnherm, 2018-02-18:
Oh yes, sorry. It was just under my nose. My bad!

closed 2018-02-18 by martin-ger