martin-ger/esp_mqtt #34: Using ESP as AP and local MQTT broker

llewmihs, 2018-04-13:
I'm looking to use the ESP as an Access Point and run an MQTT broker on the ESP on a network hosted by the ESP. I can see how to set a host as an external MQTT provider, saying running on my home computer, but how do I set the client on the ESP, as local. And what is the IP to access this through something like Node-Red. I'm trying 192.168.4.1:1883 but no luck.

Help gratefully received.

martin-ger, 2018-04-16:
Don't understand exactly - when you are running my prog as AP, you should be able to connect an MQTT client like mqtt_spy or mosquitto_sub/mosquitto_pub to port 1883 at 192.168.4.1.

llewmihs, 2018-04-16:
I thought that should be the case. When I enable the MQTT by `set mqtt_host` what should i use as the host? 192.168.4.1?
When I try accessing the broker from a client like Node-Red, it is unable to connect.
Thanks for your support here.

martin-ger, 2018-04-16:
No need to set "mqtt_host" (this is the address of a 'remote' broker). The 'local' broker is running right after startup. Just connect to the ESP's AP and with an MQTT client like mqtt_spy or mosquitto_sub/mosquitto_pub to port 1883 at 192.168.4.1 - it will work.

Go into the CLI (serial 115200 or telnet (or netcat) to port 7777) and look into "show mqtt" and you should see the MQTT client's name in the list.

closed 2018-06-08 by martin-ger