martin-ger/esp_mqtt #35: Numeric Comparisons

steverrobertson, 2018-05-08:
Martin,
First I want to say thank you thank you for such a great bit of software!  I am having difficulty with a script I am trying to create, and I cannot figure out how to do it.  I receive a value from another MQTT client into this broker.  This value is between 40.8 and 59.2.  If the value is greater than 54.5 I want to turn on the relay.  If the value is less than 51.7 I want to turn off the relay.  I get an error when I load the script, I get an error.  My script I'm having difficulty with is as follows:

on topic local $Voltage_topic
do
	if $this_data > 54.5 then
		setvar $relay_status = 1
		gpio_out 14 $relay_status
		gpio_out 13 not ($relay_status)
	else
	    if $this_data < 51.7 then
		setvar $relay_status = 0
		gpio_out 14 $relay_status
		gpio_out 13 not ($relay_status)
	    endif
	endif

	publish remote $Voltage_topic $this_data

Can you please show me what I am not understanding?  Thanks in advance!  Steve

martin-ger, 2018-05-08:
Okay, without testing it, I see two problems:

1) We don't have "<", but "51.7 > $this_data" should do the job
2) (more serious) We don't have floating (or fixed) point vals, just integers. If you are sure, that you need not more than 1 digit after the '.' you can write the following to convert it to a kind of fixed point:

// Get the integer part
setvar $int = csvstr($this_data, 0, ".")

// Get the part (after the "."), if non just 0
setvar $frac = 0 | csvstr($this_data, 1, ".")

// Combine it to 10 * the val
setvar $fixed_point = 10 * $int + substr($frac, 0, 1)

This should convert e.g.: 61.28 to 612 and 62 to 620.

As I said: not tested here, but I this you see the idea...

steverrobertson, 2018-05-08:
Such a beautiful lightweight script.  Works as described.  Thank you!  I now have a remote voltage-controlled switch!

closed 2018-05-08 by steverrobertson