martin-ger/esp_mqtt #37: Change Wifi AP name in user_config.h has no effect

miseryofdevice, 2018-06-11:
Branch: Master
Commit: 6964c0eb501769f58632ce72544b6ebc9e31f4b9
Device: Sonoff switch

According to the "Usage" section of the `README.md` the default WIFI AP name can be changed by editing file `user_config.h`.
But when I change line 9 in `user/user_config.h`, it has no effect after flashing the device. I seems like the configurations are stored on some not flashed eeprom. Is this true? How can I overwrite those configurations?

When I change e.g. printf's in file `user_main.c` I see changes. So my tool chain and flashing works.

martin-ger, 2018-06-11:
The current config is stored in flash after first boot. To change it back to the values from user_config.h enter
`reset factory`
on the command line (serial port 115200 bps) or clear the flash mem at 0xc000.

miseryofdevice, 2018-06-11:
Hi Martin,

Thanks for the quick reply.

But when I change the file `user/user_config.h` line 9-10 to e.g. 
`#define WIFI_AP_SSID         "Banana"`
`#define WIFI_AP_PASSWORD     "Banana"`
and then flash the device by
`make clean && make && make flash`
the AP name is still the same. I need to run the cmd `reset factory` on every flashed device by hand to make the AP name change? 
Furthermore the pw is still not required. What did I wrong here?

martin-ger, 2018-06-12:
Hi,

no, you do not need that on each new device. Only, if you had flashed it before with the sw with other default values or if you saved your own config, if will remember that and normal flashing of the esp_mqtt binaries doesn't overwrite these values.

If you want to make sure, that every flashed devices gets the new default, change
`#define MAGIC_NUMBER 0x015005fd`
in config_flash.h to a random other value. This marks the config sector as valid.

If you want to have a WPA-PSK pw-protected AP on init, also change in "config_flash.c":
`config->ap_open = 1;`
to 0. In this file you also find the init of all other defaults.

miseryofdevice, 2018-06-12:
Good. This works for now. I will work with you stuff for a while and maybe you will get some PRs. ;)
Great job btw! :)

closed 2018-06-13 by martin-ger