martin-ger/esp_mqtt #38: open AP with ap_ssid MyAP not available

tmurngon, 2018-06-22:
Hi there,
Your work seems great and just what I was looking for.
As I do not have the skill to build the code myself I am trying to use the pre-built binaries.

Flashing them to my NodeMCU seems to work fine (no error messages etc).
However, the 'open AP with ap_ssid MyAP' (or any other name for that matter) does not seem to show up for me to connect to.
Any guidance in this regard would be much appreciated.

Thanks.
Jan

martin-ger, 2018-06-23:
Did you try DIO mode? Try to download the whole zip archive and extract the binaries - direct download of the binaries might corrupt the files (cr-lf). What does the serial line at 115200 bps show after flashing?

tmurngon, 2018-06-24:
Excellent! I managed to connect.
The problem seems to have been the beed for DIO. I thought I had this set before, but my flashing software had a different idea...
Thank you so much for your help!

closed 2018-06-24 by martin-ger