martin-ger/esp_mqtt #39: UI for setting up AP

tmurngon, 2018-06-25:
Hi Martin,
In the README I see:
"While the user interface looks similar to my esp_wifi_repeater at https://github.com/martin-ger/esp_wifi_repeater", which leads me to believe that there is a GUI available at http://192.168.4.1 )or somewhere similar) which can be used to do the initial setup.

However when visiting 192.168.4.1 (or http://192.168.4.0) I simply get a timeout.

Am I wrong in my assumption, or am I accessing the UI incorrectly perhaps?

BTW the Telnet interface etc is all working fine.

Thanks.

martin-ger, 2018-06-25:
No, there is no Web-based GUI for the broker. This would require additional mem, which is already short for a number of clients.

The standard config interface is the command line at port 7777 or via the serial port at 115200.

tmurngon, 2018-06-25:
Thanks Martin,

I was just asking because one of the first concerns when handing a device over to a third party user is how they will set the device up without much technical skill.
I understand the memory issue though.