martin-ger/esp_mqtt #42: loop

msillano, 2018-07-03:
Testing a recursive script, Sonoff goes on loop. Using the serial console I get:

```
. . . . 
bcn 100

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 30828, room 16
tail 12
chksum 0x40
ho 0 tail 12 room 4
load 0x3ffe8000, len 2124, room 12
tail 0
chksum 0x12
load 0x3ffe8850, len 10452, room 8
tail 12
chksum 0x86
csum 0x86
▒▒▒▒x▒n▒▒>r▒▒▒n▒b
▒l▒▒l`▒▒▒▒▒▒▒b▒r▒
                 n▒c▒8▒▒
r▒▒▒▒8  ▒▒

uMQTT Broker V2.0.4 starting

Config found and loaded
Starting Console TCP Server on port 7777
Max number of TCP clients: 15
mode : sta(ec:fa:bc:06:a7:a2) + softAP(ee:fa:bc:06:a7:a2)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
. . . 
```
I cannot inser any command !.
I also re-flashed the Sonoff (using last bin files) but it stiil loops :(
Any idea?
I'm trying to do iterations using  'on topic' as recursive function.
So it is possible that the script (wrong) goes in infinite loop, giving a stack error.
But flashing Sonoff don't clear the script?

Best regards
Marco

martin-ger, 2018-07-03:
Recursion is not supported (only one level of "on" statements). Perhaps I should have a view into the script.

Flashing the binary doesn't delete the script. If you want to, do this:
`esptool.py --port /dev/ttyUSB0 erase_flash`

msillano, 2018-07-04:
A language, to be "Turing complete" must have 3 structures: sequences, selections (if, or case, - all are equivalents) and iteractions (while or do or for - all are equivalents). The MQTT script is missing iterations.  Iterations can ben done also with function recursion, but the script do not allows recursion. So it is important to add some form of iteration:.
The simplest form is maybe  ' while  <condition> do <expressions> endwhile'.

Why I need it?
I try to give you the idea (test code not whit me now, I can send it you later):  a timer, doing  tasks at any time: ex: @08:25 (long 3 min), @10:30 (long 12 min), @12:07 (long 4 min), etc....  
A json structure  (in @X) keeps in arrays the  start times (ordered) and the durations:.
I can access it using json_parse()  and an  index... 

The problem is at the start:  if Sonoff starts at 09:10, I  must set the alarm to 10:30:. To do this I must to iterate over the full array, to find the first start time bigger than 09:10: no alternatives, no way.

To give you an idea of the application, see  https://github.com/msillano/sonoff_watering  (work in progress not full finished), This poject is simpler. because it do not need iterations: times are only 2, T1 and T2,  but it is very  near at the new one.

Martin, thanks for your time.
Regards

martin-ger, 2018-07-04:
Marco,

as a computer scientist I was aware, that the language is not "Turing complete". Actually, it was originally never intended to be. A downside of being "Turing complete" is the "Halting problem" or even less, the ability to code arbitrary long loops. I was thinking of small action-response clauses, that cannot block the CPU...

For more sophisticated apps I thought using the uMQTTBroker lib with an Arduino sketch would be a generic solution.

Nevertheless, I will think about a "while" loop...

Cheers,
Martin

martin-ger, 2018-07-04:
Just added a "while do" to the language... ;-)

msillano, 2018-07-05:
hahaha wonderful !
Thanks. i will try it soon.