martin-ger/esp_mqtt #43: Script how to subscribe to # ?

j0hncc, 2018-08-07:
In my script I can subscribe and receive specific events no problem.
But if I try to receive all events first I get script error because of #  "Error (number of hexdigits must be multiple of 2) at >># on topic local "#...
If I put # in double or single quotes it accepts the script but no topics are matched and printed to serial_out.  Also tried \#.

on init
do
   subscribe local #

on topic local #
do
  serial_out "Received: " | $this_topic | " , " | $this_data

Is there some other trick to subscribe to # ?

Thanks!

martin-ger, 2018-08-07:
Have auch look at the bridge sample. Thus should work.

j0hncc, 2018-08-07:
Sorry, my mistake.  I didn't realize that after reloading a new script, I needed to reset for it to take effect...
The proper format is indeed subscribe local "#"

closed 2018-08-07 by j0hncc