martin-ger/esp_mqtt #46: Flashing the firmware binaries to a sonoff

HoerMirAuf, 2018-11-08:
Hey there ....

i was downloading the prebuild binaries from the firmware folder:
0x00000
0x10000
I'm little bit confused...
Have i flash the 0x10000 to 0x02000 like in the picture or have i flash it to 0x10000?
I tryed out both Variants, with all the settings like the manual Picture from ESP Download Tool, but nothing will happen after flashing....
Also tryed out with Flash Size 8Mb and 32Mb and Baud 15200 and 220400.

What i'm doing wrong?

martin-ger, 2018-11-08:
Sorry, the image is linked incorrectly - 0x10000 is correct.

HoerMirAuf, 2018-11-08:
Hi martin-ger.

Thank you for the Information.
I flashed both files, both with Download Tool (Windows) and esptool.py (linux) to my sonoff 1MB Device.
But nothing happens after flashing. No AP, no seriall output, nothing.

Both files togehter seems a little bit small, just 108k. Everyone 54k. Are the binaries correct? Is it neccessary to flash any other binarie first?

martin-ger, 2018-11-08:
The file sizes are wrong. Should be: 29KB and 222KB. Download the complete repo as zip and extract the binary files. Works best...

HoerMirAuf, 2018-11-09:
Yep ! Works fine with the binaries from the zip folder.

Thanks a lot :)

closed 2018-11-09 by HoerMirAuf