martin-ger/esp_mqtt #47: set ntp publish interval

nopnop2002, 2018-12-26:
Thank you for useful Firmware.

After NTP sync has been completed successfully once, 
the local time will be published **every second** under the topic "$SYS/broker/time" in the format "hh:mm:ss".

I want to set publish interval.

set ntp_local_interval 600

the local time will be published **every 10 minute** under the topic "$SYS/broker/time" in the format "hh:mm:ss".

martin-ger, 2018-12-26:
This is quite a specific request. This could be done in the code, but in order to keep that somewhat generic, I would suggest to use a script with two timers and two alarms.

nopnop2002, 2018-12-26:
Thank you very much for your answer.
I'll use a script.

martin-ger, 2018-12-27:
As it might be soewhat tricky, here is a script that does the job:
```
% Example for some regular timers

% Config params, overwrite any previous settings from the commandline
config ntp_server	1.pool.ntp.org
config ntp_timezone	1

% Now the initialization, this is done once after booting
on init
do
    setvar $t_prefix = "$SYS/broker/time"
    setvar $tt1 = $t_prefix | "1"
    setvar $tt2 = $t_prefix | "2"
    setvar $tt3 = $t_prefix | "3"
    setvar $tt4 = $t_prefix | "4"

    % Starts, when time is synced
    subscribe local $t_prefix

% Now the events, checked whenever something happens

on topic local $t_prefix
do
    % Wait for sync
    unsubscribe local $t_prefix
    settimer 1 1000
	setalarm 1 "12:00:00"
	setalarm 2 "00:00:00"

on timer 1
do
    % Check for full minute
    if (csvstr($timestamp,2,":") = "00") then
	    publish local $tt1 $timestamp
    endif
    % Check for full hour
    if (csvstr($timestamp,1,":") = "00") then
	    publish local $tt2 $timestamp
    endif
	settimer 1 1000

on alarm 1
do
	publish local $tt3 "12:00:00"

on alarm 2
do
	publish local $tt4 "00:00:00"
```

nopnop2002, 2018-12-27:
I worried about this 

the local time will be published every 10 minute
8:10 8:20 8:30 8:40 .......

the local time will be published every 60 minute
9:00 10:00 11:00 12:00 ......

Thank you.

martin-ger, 2018-12-28:
Not what you requested for? Do you want an event exactly 10 min after sync?

nopnop2002, 2018-12-28:
Your script works perfectly.

`	Wait for WiFi....WiFi connected
	IP address: 192.168.4.2
	subnetMask: 255.255.255.0
	gatewayIP: 192.168.4.1
	Connect to 192.168.4.1
	clientid=ESP8266-cb120b
	Attempting MQTT connection...connected as subscriber
	Message arrived [$SYS/broker/time1] 07:00:00
	Message arrived [$SYS/broker/time2] 07:00:00
	Message arrived [$SYS/broker/time1] 07:10:00
	Message arrived [$SYS/broker/time1] 07:20:00
	Message arrived [$SYS/broker/time1] 07:30:00
	Message arrived [$SYS/broker/time1] 07:40:00
	Message arrived [$SYS/broker/time1] 07:50:00
	Message arrived [$SYS/broker/time1] 08:00:00
	Message arrived [$SYS/broker/time2] 08:00:00
`

martin-ger, 2018-12-29:
Q1: That't correct, the script and the WiFi-Settings are stored independently from the code. Clear flash (or write any file to 0x00000 - at least 64 KB). This will reset the settings as well.

Q2: Don't know yet...

Happy New Year!