martin-ger/esp_mqtt #48: Request: "MQTT client user" too short for my cloud broker

abaubi, 2018-12-31:
Hi Martin,

First of all congratulate you for your great job.

I have a request for you. I can't publish or subscribe anything from my cloud broker because my "MQTT client user" is used as the TOKEN to connect to it and it is too short, in fact, it is too short for just 3 characters.

I really appeciate you could increase the size of this configuration variable in next firmware update.

Thanks and keep it up!

CMD>show
Version V2.0.7 (build: Thu Jul 19 08:06:57 2018)
STA: SSID:TP-LINK_XXXXXX PW:XXXXXXXXXX
AP:  SSID:MyAP PW:none [open] IP:192.168.4.1/24
Network console access on port 7777 (mode 3)
MQTT broker max. subscription: 30
MQTT broker max. retained messages: 30
MQTT client enabled
MQTT client host: XX.XX.XX.XX
MQTT client port: 1883
MQTT client user: XXX-ur1EjikpRuXXxNvd1Vua1r11aYh (3 characters are missing from my conf!!!)
MQTT client password:
MQTT client id: ESPBroker_23d0ad
MQTT SSL: off
NTP server: 1.pool.ntp.org (interval: 300 s, tz: 0)
Clock speed: 80
Serial bitrate: 115200

abaubi, 2019-01-03:
Hi martin

I tested the new version, as you can see, mqtt_user and mqtt_password are still truncated. MQTT client is still disconnected.

I stay tunned for next version.

Regards

CMD>show
Version V2.0.8 (build: Wed Jan  2 07:52:51 2019)
STA: SSID:XX-XXXX_XXXXXX PW:XXXXXXXXXX
AP:  SSID:MyAP PW:none [open] [disabled] IP:192.168.4.1/24
Network console access on port 7777 (mode 3)
MQTT broker max. subscription: 30
MQTT broker max. retained messages: 30
MQTT client enabled
MQTT client host: XXXXXXXXXXXXX
MQTT client port: 1883
MQTT client user: XXX-ur1EjikpRuXXxNvd1Vua1r11aYh ->(Still truncated)
MQTT client password: XXX8545XXX204ea7bXXXfe42ebXXXf6e ->(Still truncated)
MQTT client id: ESPBroker_07b533
MQTT SSL: off
NTP server: 1.pool.ntp.org (interval: 300 s, tz: 0)
Clock speed: 80
Serial bitrate: 115200

martin-ger, 2019-01-03:
Sorry, you were asking for the mqtt_id. You may fix it yourself easily be recompiling it with changed constants in user_config.h. We do it in the next version...