martin-ger/esp_mqtt #49: Request: "MQTT client user" too short for my cloud broker #48

abaubi, 2019-01-03:
Hi martin

I tested the new version, as you can see, mqtt_user and mqtt_password are still truncated. MQTT client is still disconnected.

I stay tunned for next version.

Regards

CMD>show
Version V2.0.8 (build: Wed Jan  2 07:52:51 2019)
STA: SSID:XX-XXXX_XXXXXX PW:XXXXXXXXXX
AP:  SSID:MyAP PW:none [open] [disabled] IP:192.168.4.1/24
Network console access on port 7777 (mode 3)
MQTT broker max. subscription: 30
MQTT broker max. retained messages: 30
MQTT client enabled
MQTT client host: XXXXXXXXXXXXX
MQTT client port: 1883
MQTT client user: XXX-ur1EjikpRuXXxNvd1Vua1r11aYh ->(Still truncated)
MQTT client password: XXX8545XXX204ea7bXXXfe42ebXXXf6e ->(Still truncated)
MQTT client id: ESPBroker_07b533
MQTT SSL: off
NTP server: 1.pool.ntp.org (interval: 300 s, tz: 0)
Clock speed: 80
Serial bitrate: 115200

_Originally posted by @abaubi in https://github.com/martin-ger/esp_mqtt/issues/48#issuecomment-451208322_

pablopescador, 2019-06-18:
Hi Martin,
grüezi aus München, 
Wie kann ich die ESP8266 (mit esp_mqtt) mit  ADAFRUIT.IO binden. Ich habe verschiedene form für die Verbindung aber habe ich bis jetzt keine erfolgt.
Vielen dank im Voraus,
SG,
Pablo.

Goltergaul, 2022-10-31:
Hi, I have the same problem, but with the hostname of the external mqtt server. i got one from hivemq, which is pretty long. unfortunately it gets truncated. Is there any way around this?