martin-ger/esp_mqtt #55: GPIO1 on startup

matejb123, 2020-04-17:
Hi,
I want to use GPIO1 as an output, but if I have the peripheral connected, the ESP doesn't boot.
If I connect the peripheral after boot, then everything works as intended.
Am I missing something? I don't need serial.
Thanks for good work and best regards.
Matej

justoke, 2020-04-17:
GPIO1 is HIGH at boot, so if you have something connected that is pulling it low you will likely get unexpected behaviour or failure to boot. You could try ensuring that the boot state of whatever you connect there is HIGH. You can still only use GPIO1 as an output. There should be no Serial.begin, print anywhere in the setup() or loop() methods.

If you are using an ESP32 module - such as [this](https://www.ebay.co.uk/i/254253057977?chn=ps&norover=1&mkevt=1&mkrid=710-134428-41853-0&mkcid=2&itemid=254253057977&targetid=878706529565&device=c&mktype=pla&googleloc=9044969&poi=&campaignid=9794911188&mkgroupid=101625062844&rlsatarget=pla-878706529565&abcId=1139126&merchantid=118942247&gclid=EAIaIQobChMI8s_l1afw6AIVSbTtCh3yngmmEAQYAiABEgJFVvD_BwE) then you will note that GPIO1 pin  is wired to the TX pin for USB to TTL chip for the purposes of programming the board. That is why Serial is reserved for this purpose even if you don't use it. There are ways to disable this Serial port but then you wouldn't be able to program the ESP32.

matejb123, 2020-04-18:
Hi,
Thanks for reply. Hm... I'm trying to power a sensor with polling, i.e. I don't want the sensor be powered all the time via Vcc. The sensor is basically a comparator but I suspect the blocking capacitor on its Vcc pin is pulling gpio1 down for a short period of time.
Has anyone of you run into this problem? Would a lower value capacitor help?
Thanks and best regards,
Matej

justoke, 2020-04-18:
You could use a P-Channel MOSFET to switch the sensor on. It would be OFF during the boot sequence, in setup() keep it HIGH until you want to poll the sensor. This should negate the influence the capacitor might have on your circuit.

matejb123, 2020-04-25:
Thank you very much!
Regards,
Matej

closed 2020-04-25 by matejb123