martin-ger/esp_mqtt #59: Subscribing based on partial string match.

jampez77, 2020-07-19:
I love this project as it's helping make sure that my Router isn't dropping the many device connected to it.

I'm using this as an MQTT bridge and would like to know if it's possible to subscribe to a topic by using a substring.

So for example I would like the relay all MQTT message from topics containing the word `middleFloor` or `groundFloor` 

I know there is the `substr` method that would allow something similar but it's not ideal as I would need to know all the required string lengths before hand.

martin-ger, 2020-07-19:
Sorry, this is not possible. According to the MQTT spec, you can use wildcards, but only for complete levels. If possible, try to arrange topics like this: /sensor/groundFloor/s1, /sensor/groundFloor/s2...
See also here: https://www.hivemq.com/blog/mqtt-essentials-part-5-mqtt-topics-best-practices/

jampez77, 2020-07-20:
Thank you! that link was a huge help! I've changed my topics and it's working as desired.

closed 2020-07-20 by jampez77