martin-ger/esp_mqtt #64: ESP постоянно перезагружается.

NOV62, 2020-12-01:
Dear Martin, there is a new problem. I updated the esp_mqtt firmware to the latest one, from 20.02.2020, now my ESP8266 module is constantly rebooting. Tested on 3 modules, ESP-01, ESP-01S and ESP-12. Flashed in QIO mode, and in DIO mode, here is a screenshot:
[image]

Unfortunately, I didn't save the old bin files that ESP worked with stably. Could you post the old version?

martin-ger, 2020-12-01:
I don't understand why. Did you download the zip-archive and extract the binaries - this works better than downloading the binaries directly via the browser.

In any case you can get the old binaries by going back in the repos history. All versions contain the binaries.

NOV62, 2020-12-01:
Sorry, I don't understand how to do this. The Releases tab contains only archives with code, no binaries. Where can I find repos history?

martin-ger, 2020-12-01:
Choose any version here: https://github.com/martin-ger/esp_mqtt/network

NOV62, 2020-12-01:
Ещё раз извиняюсь, нашёл. Я новичек на GitHub, поэтому ещё очень мало знаю. Спасибо!

closed 2020-12-02 by NOV62