martin-ger/esp_mqtt #66: MQTT translation

mcsabi, 2021-06-22:
First of all thank you for the excellent project. So far it works flawlessly. I have a noob question as I am not a programmer. 
I have a thermostat scheduler that sends an MQTT message when the relay should turn on/off. A remote tasmotized device should receive the message and turn on/off its relay accordingly. The message that is being published "cmnd/heater/relay/1 1" however the tasmota device would expect "cmnd/heater/POWER1 1". On the publishing device I cannot change the "/relay/1" part of the topic and on the tasmota I cannot change the "POWER1" part of the topic. Can I somehow do a translation in the uMQTTBroker so that the MQTT message would be delivered? or my whole thinking is wrong?
Many thanks in advance!

martin-ger, 2021-06-22:
Can be done easily using a script like this (didn't check it, but should work): 

```
on init
	subscribe local "cmnd/heater/relay/1"

on topic local "cmnd/heater/relay/1"
do
	publish local "cmnd/heater/POWER1 $this_data
```

mcsabi, 2021-06-22:
Thank you so much, will try soon!

mcsabi, 2021-06-23:
I have tested the script you suggested, but unfortunately I could not make it work. I have tried:

on init
	subscribe local "cmnd/heater/relay/1"

on topic local "cmnd/heater/relay/1"
do
	publish local "cmnd/heater/POWER1 $this_data

also tried with one more "do", like:
on init
do
	subscribe local "cmnd/heater/relay/1"

on topic local "cmnd/heater/relay/1"
do
	publish local "cmnd/heater/POWER1 $this_data

I have used Netcat from Windoes Nc64.exe to upload the script. It seems to be fine as it shows the script with the "show script" command, however after reboot it cannot connect to the network. My test ssid is "24" and this message is repeated on serial port:
scandone
no 24 found, reconnect after 1s
disconnect from ssid 24, reason 201
reconnect
scandone

After factory reset all works fine until I upload the script and reboot. BTW the script upload takes a long time for some reason.
Any hints would be appreciated.
Thank you

martin-ger, 2021-06-23:
This works for me:
```
config speed		160

on init
do
	subscribe local "cmnd/heater/relay/1"

on topic local "cmnd/heater/relay/1"
do
	publish local "cmnd/heater/POWER1" $this_data
```

For some reason that I don't know yet, it has these problems you reported when there is no config line in the beginning. BTW: You can ^C the netcat, the script is still loaded.

mcsabi, 2021-06-23:
Works like a charm! I do not understand why can it connect to my wifi with the "config speed 160" in the script and cannot w/o, but  it works. Thank you again